neu-rah/ArduinoMenu #407: Raspberry Pi PICO RP2040

hicksan, 2023-02-24:
Can this library be used with RP2040? Apparently lots of Arduino libraries have been ported already. It would be nice to have a menu interface consistent across devices.

datorvictor, 2023-09-30:
works fine on my Pi PICO

mikejr83, 2024-01-20:
I'm also having the same issue. I cannot get this to build with my platform.io setup:

Getting this at the end of a ton  of warnings (and a few other errors):
```
*** [.pio\build\pico\lib8d7\ArduinoMenu library\menuIo.cpp.o] Error 1
*** [.pio\build\pico\lib8d7\ArduinoMenu library\nav.cpp.o] Error 1
*** [.pio\build\pico\src\main.cpp.o] Error 1
```

Here's the `platformio.ini` file:

```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
monitor_speed = 115200
lib_deps =
  neu-rah/ArduinoMenu library @ 4.21.3
```

@datorvictor Would you mind sharing a basic example?

mikejr83, 2024-01-21:
@hicksan I was able to get this to work for my pico. I have an easy way to do this iff you're using platformio with VSCode. Look in `./.pio/libdeps/pico/Arduino Menu library/src/macros.h`. Head to line 55. Comment out lines 55-57.

mikejr83, 2024-01-22:
Digging deeper I found you can add a build flag `-D USE_PGM`. Forces the storage into flash memory. I wish I new how to fix those lines to detect the pico and not use them somehow.

datorvictor, 2024-01-23:
I do not remember doing anythin special. I use VS Code with arduino.

This is propably relevant from my code:

```
#include <Arduino.h>
#include <U8x8lib.h>
#include <U8g2lib.h>

#include <Wire.h>
#include <WiFi.h>
#include <menu.h>
#include <menuIO/serialOut.h>
#include <menuIO/serialIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/U8x8Out.h>
#include <Keypad_MC17.h>
#include <menuIO/keypadIn.h>

.....

U8X8_SSD1322_NHD_256X64_4W_SW_SPI u8x8(/* clock=*/_PIN_SCK_, /* data=*/_PIN_MOSI_, /* cs=*/_PIN_CS_, /* dc=*/_PIN_DC_, /* reset=*/_PIN_RST_);

...
serialIn serial(Serial);
keypadIn kpad(customKeypad);

MENU_INPUTS(in, &kpad);  //, &serial);

#define MAX_DEPTH 2

MENU_OUTPUTS(out, MAX_DEPTH, SERIAL_OUT(Serial), U8X8_OUT(u8x8, { 0, 0, 28, 8 }));
NAVROOT(nav, mainMenu, MAX_DEPTH, in, out);`
```

my arduino.json:
`{
    "configuration": "flash=2097152_1048576,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,wificountry=worldwide,ipbtstack=ipv4only,uploadmethod=default",
    "board": "rp2040:rp2040:rpipicow",
    "sketch": "PicoKBD.ino"
}`

some of the 10000 defines generated automatically:
`"ARDUINO_VARIANT=\"rpipicow\"",
                "TARGET_RP2040",
                "PICO_FLASH_SIZE_BYTES=2097152",
                "ARM_MATH_CM0_FAMILY",
                "ARM_MATH_CM0_PLUS",
                "F_CPU=133000000L",
                "ARDUINO=10607",
                "ARDUINO_RASPBERRY_PI_PICO_W",
                "BOARD_NAME=\"RASPBERRY_PI_PICO_W\"",
                "ARDUINO_ARCH_RP2040",`