neu-rah/ArduinoMenu #409: Preventing the rotary encoder to cycle through

dhysuiej, 2023-03-12:
I want a menu or numeric field to stop changing its value one it reaches the upper or lower limit in the range. Is that posible with the latest released version or would I need to change the corresponding classes?

keldonin, 2023-03-21:
For both `MENU` and `FIELD`, use `Menu::noStyle` instead of `Menu::wrapStyle` for the last argument.

In the following example, both the menu and the "LED" field are constrained and do not wrap around.

```c++
MENU(mainMenu, "configuration", Menu::doNothing, Menu::noEvent, Menu::noStyle
  ,FIELD(pwmled,"LED","", 0,255,10,1,Menu::doNothing, Menu::noEvent, Menu::noStyle)
  ,EXIT("<Back")
);
```

It works for me, with the current library release.

dhysuiej, 2023-03-21:
@keldonin thanks a lot.

closed 2023-03-21 by dhysuiej