neu-rah/ArduinoMenu #433: Support for Initialization Outside of Main

mikejr83, 2024-01-20:
I'm trying to get this library set-up so that I can initialize a rather complex menu structure in a separate file.  How can I do this? It doesn't seem possible that I can return nav:

MenuConfig.cpp
```c++
Menu::navRoot CreateMenu()
{
  // create menu structure
  NAVROOT(nav, mainMenu, MAX_DEPTH, in, out);

  return nav;
}
```

main.cpp
```c++

Menu::navRoot nav = CreateMenu();

void setup(void) 
{
  // set up Serial, etc.
}

void loop(void)
{
  nav.poll();
}
```

I've tried several ways, like passing references, etc. The program seems to crash in the loop.

Is there a way to do this or do am I forced to have a gigantic `main.cpp` file?!

mikejr83, 2024-01-21:
I found issue #293. This gives a great start, but I'm not sure how to do this within a helper function? The OP of that issue has an unresolved question on how to provide the return type. I've been struggling with trying to wrap my head on how to get the references in and out of a call to the function. I'm a C# dev so grasping how pointers and references work in this situation such that anything created locally in the function isn't destroyed is killing me.

neu-rah, 2024-01-21:
not sure if this helps, if dealing with value fields, they are hooked to your variables, so the values are already on the variables...
but if you still want you can capture the events, see this example https://github.com/neu-rah/ArduinoMenu/blob/master/examples/handlers/handlers/handlers.ino