neu-rah/ArduinoMenu #444: ClickEncoder.h: No such file or directory

shaneodd, 2025-03-12:
# Issue/Error
`src/menuIO/clickEncoderIn.h:34:16: fatal error: ClickEncoder.h: No such file or directory`

I am following this guide for using a 3d printer screen and controls. https://www.bjonnh.net/article/20221126_mini_12864/

When I do the include `#include <menuIO/clickEncoderIn.h>` the above error shows up. Not exactly sure what are my next steps are, I have tried different versions of the plugin with no change in this error. I am guessing there is an additional package I may need to install since I see other s reference the same clickEncoderIn file.

# System info
Linux Mint
Arduino 2.3.4 AppImage

neu-rah, 2025-03-13:
using Arduino Ide? PlatformIO? ...?

shaneodd, 2025-03-13:
I have the arduino ide 2.3.4 specifically the app image version.

shaneodd, 2025-03-19:
Looks like there is something with the include in that file. I tried it on a windows 11 box with the latest ide and same error as the linux box.

neu-rah, 2025-03-21:
are you including the library before?
```c++
#include <ClickEncoder.h>
#include <menuIO/clickEncoderIn.h>
```

shaneodd, 2025-03-22:
In clickEncoderIn.h the file ClickEncoder.h is being included. 

https://github.com/neu-rah/ArduinoMenu/blob/ca689063029e414a86109e9329b774ec14f59726/src/menuIO/clickEncoderIn.h#L34

shaneodd, 2025-03-23:
> are you including the library before?
> 
> #include <ClickEncoder.h>
> #include <menuIO/clickEncoderIn.h>

I have also tried this as well. Is clickencoder pulled in from another library?

Commenting it out from the arduinomenu library file i get this error
`Compilation error: 'ClickEncoder' does not name a type`

shaneodd, 2025-03-23:
Actually answered my own question, https://github.com/0xPIT/encoder is what is needed. I would maybe throw a note on the examples where this is included. If people in the future run into this and they miss that on the main docs that would serve as another reminder since there is a LOT of info on the main read me.

neu-rah, 2025-03-27:
Thank you.
The wiki on this project is open for edit and PRs are very welcome